ecc1/gnarl #2: GNARL battery-test with Loop

cfaagaard, 2019-11-08:
hi,

I have made test with the following setup:
- Loop (current dev)
- iPhone 5s (ios 12.4)
- Medtronic pump 522
- CGM (SG from Dexcom share)
- GNARL-code from 11.9.19
- GNARL-hardware (https://www.banggood.com/2Pcs-Wemos-TTGO-LORA32-868915Mhz-ESP32-LoRa-OLED-0_96-Inch-Blue-Display-p-1239769.html)
- Battery 850 mah

[image]

Test:
- 10 hours charging of battery on GNARL-hardware
- Running Loop connected to Medtronic Pump and GNARL
- Did some bolus, suspend pump, overrides, changed settings
- Used Nightscout to collect data via Loop (version 0.12.5)

Result:
I did the test two times and both tests showed a battery-lifetime just around 18 hours.
This is compared to the Riley link which have 24-30 hours battey-lifetime and 2 hours recharging.

 Observations/suggestions:
- Wifi: I can see that wifi-module is included in the code. In the GNARL/Loop scenario wifi is not needed. If the wifi-hardware is turned on, turning this off will increase batterylifetime.

- No Bluetooth-connected indication on the GNARL (like Rileys green light).

- GNARL sleep-modes: Maybe we should look more into this. But I do not think it possible to wake it up from a bluetooth signal.

trixing, 2019-12-14:
How did you measure the recharging time? From the ttgo datasheet [1] it uses a TP4054 charging circuit [2] with a 2k Rprog resistor (R23), which should be a bit less than 400mA charging current. For a 850mAh battery the charging time should be less than 2 hours.

That said: I measured R23 on my board (the outer 5pin IC next to the USB plug, 0402 resistor) and it has a value of 10k which equates closer to 100 mA charge current and could explain your 10 hour time you are seeing if your battery was really empty.  Might be worth trying solder a 2k resistor in instead (or on top).

[1] https://github.com/LilyGO/TTGO-LORA32-V2.0/blob/master/schematic1in6.pdf
[2] e.g. https://www.analog.com/media/en/technical-documentation/data-sheets/405442xf.pdf

trixing, 2019-12-14:
I just realized I was looking at the V2 schematic. So what I wrote above assumes that the charging circuit is the same IC (I cannot read the markings).

cfaagaard, 2019-12-16:
@trixing I did not measure the recharging time. It was ecc1 who said that it was about 8,5 hours (https://gitter.im/openaps/hardware-dev?at=5dc08b5ffb4dab784a645797 )

trixing, 2020-07-03:
I finally got around testing the theory.  With a fine solder tip, I replaced the 10k resistor with a 0402 2K.  While I don't have a USB current meter readily available, the voltage rises noticeably faster and goes from 3.3V to 4.2V in ~4 hours on a 1200mAh Lipo. ymmv.

Btw, the board itself already consumes ~50mA.  That means the battery in the default state only charges with 50mA, that explains why an 850mAh battery takes >>10 hours to charge.

[image]