ecc1/gnarl #4: Problems connecting to the AndroidAps.

Lexsus, 2019-12-19:
In the beginning, a successful connection via bluetooth to the AAPS occurs. But then after some time (10-20 minutes) the commands from the AAPS to the gnarl do not sent. After the reboot, the connection is restored.

p.s I use heltec wifi lora module, I do not think that the problem is in this module.

diskaril, 2020-02-14:
Have you figured this out? I also have the Heltec wb32la V2 board which is working fine, the performance is very similar to my rileylink on AAPS.

Lexsus, 2020-02-14:
> 
> 
> Have you figured this out? I also have the Heltec wb32la V2 board which is working fine, the performance is very similar to my rileylink on AAPS.

I have problems with connection.

ecc1, 2020-02-14:
No, I don't really have the setup to debug this much further, and I don't use it "in production" myself. I'll accept patches if you or others find a fix.

diskaril, 2020-02-16:
> > Have you figured this out? I also have the Heltec wb32la V2 board which is working fine, the performance is very similar to my rileylink on AAPS.
> 
> I have problems with connection.

Try this steps, it may work for your board. 
      https://gitter.im/openaps/hardware-dev?at=5e479dcd48ca2e297451bb0f