ecc1/gnarl #11: Cannot set a longer BLE name than 5 characters

trixing, 2020-05-18:
First off, custom name is broken, but you probably know that. I have a fix for that upcoming.  I have that 99% working, only that I cannot set a name longer than "GNARL". Which is - what?

If I call ble_svc_gap_device_name_set with a name longer than "GNARL" the function ble_gap_adv_set_fields returns error 3 and everything restarts due to the assertion.

Setting the name to "GNARO" works, "GNARL2" does not work.  According to the spec, this should be a 130 character field.

This is with just this one line changed, and not any of my other custom name changes.

Any idea where to even start searching for the culprit?

ecc1, 2020-05-19:
There's a hard limit on the size of advertising packets. To get a longer name, we'd have to move to a follow-on scan response with the additional info that doesn't fit in the advertisement. I think I was just too lazy to implement that.

trixing, 2020-05-21:
The SDK menuconfig says 31 bytes, and this example code does not seem to have a particular configuration either (it matches the gnarl code pretty closely in fact)

https://github.com/espressif/esp-nimble/blob/nimble-1.2.0-idf/apps/bleprph/src/main.c

trixing, 2020-05-21:
Never mind, got it. Pull request forthcoming.  This is a pretty -- interesting -- library...

trixing, 2020-05-21:
See https://github.com/ecc1/gnarl/pull/15

ecc1, 2020-05-24:
Closed with f56d542e0c5518fd006dbb917e7ec72775b594e0

closed 2020-05-24 by ecc1