ecc1/gnarl #19: Inaccessible pump

toniteh78, 2021-05-22:
Hello, I have 3 medtronic pumps. One is 712 which works perfectly with Gnarl. The other two are 722 which connect and the connection to the pump constantly breaks down. One is CA (Canadian) the other is WW (the rest). Can you tell me  where could the problem be?

ecc1, 2021-05-22:
Your WW pumps use the 868 MHz band. Did you change PUMP_FREQUENCY in include/pump_config.h appropriately?

toniteh78, 2021-05-22:
I have already done it, the one that is 712 is also WW and works with it perfectly. But with the other two 722, it simply shows an inaccessible pump in AAPS.

ecc1, 2021-05-22:
The pump radios have a fairly narrow usable frequency range, and they can vary by a few hundred kHz from pump to pump. So some trial and error might be needed. 

You can also try building the mmtune application in the gnarl tree to search for the best frequency.

mountrcg, 2021-06-03:
@toniteh78 would this one work for omnipods? 
https://de.banggood.com/LILYGO-TTGO-T-Beam-ESP32-433-or-868-or-915-or-923Mhz-V1_1-WiFi-Wireless-bluetooth-Module-GPS-NEO-6M-SMA-LORA32-18650-Battery-Holder-With-OLED-p-1545070.html?cur_warehouse=USA&ID=513922

ecc1, 2021-06-03:
There is no support for omnipods. I think the rfm95 radio is capable of communicating with them, but someone would have to write the code.