ecc1/gnarl #21: GNARL disconnecting from Loop every 5 minutes

diabeticp, 2021-11-01:
Hi. I have installed GNARL on my TTGO v1 board and it just disconnects from Loop every 5 minutes. When I try to reconnect GNARL with Loop, it just gives an error message like "Device Error. Unknown characteristic: C842E849-5028-42E2-867C-016ADA-DA9155: The RileyLink was temporarily disconnected. Make sure the device is nearby, and the issue should resolve automatically". And when I try to reconnect it again, it just reconnects. And I need to reconnect GNARL with Loop every 5 minutes, otherwise it doesn't work. How can I fix it?

Maybe it will help, idk:
1. iPhone 12 mini, iOS 14.7.1
2. Latest Loop build, built with Xcode 12.5.1
3. ESP-IDF v4.3 
4. Latest GNARL build
5. Medtronic Paradigm Real-Time (722, World-Wide, 868Mhz)
[image]

[image]

szymjaw, 2022-03-07:
I have the same problem, with  TTGO v1 board without display.

    1.     iPhone SE 2020, iOS from 15.1 to 15.3.1
    2.     Latest Loop build with Xcode 13.2.1
    3.     ESP-IDF  v4.3 and v4.4
    4.     Latest GNARL build
    5.     Medtronic Paradigm Real-Time (722, World-Wide, 868Mhz)

trixing, 2022-03-07:
Are you using multiple devices by any chance? https://github.com/ecc1/gnarl/commit/cce3d24db2213d42e490335cdb40e1992581775f comes to mind if no commands are sent through GNARL it will reset.

trixing, 2022-03-07:
You could try reverting the commit (ie. disable the watchdog) to narrow down the problem.

szymjaw, 2022-03-23:
Unfortunately it didn't help .