ecc1/gnarl #22: having trouble with the make command

cmdillon77, 2022-05-04:
i'm having trouble withthe make command. If i download the zip file and unzip i can make the project folder. but cannot succesfully run idf.py build. if i clone the git I get a make error but the idf.py build cammand runs but stops compiling at 792/815 so i don't know what i am doing wrong.
any help would be great
[image]

ecc1, 2022-05-04:
Building GNARL has only been tested on Linux.

closed 2022-05-04 by ecc1

cmdillon77, 2022-05-06:
Thanks Eric. I'll try the Ubuntu on windows, or a raspberry pi running debian as my Linux machine to make the file to flash.Sent from Android device