ecc1/gnarl #23: GNARL failed to start with GPIO isr service is not installed

Andre1ko, 2022-10-24:
Hello
There is an issue with GNARL code running on TTGO version 1.
Booting of code failed with:
E (359) gpio: gpio_isr_handler_add(465): GPIO isr service is not installed, call gpio_install_isr_service() first.

IDF version - ESP-IDF v4.4.2-386-gee882708f8

Andre1ko, 2022-11-23:
Hello
Sorry for a long delay

The same situation with v4.3 of ESP-IDF
D (355) GNARL: enable watchdog with a timeout of 300 seconds
I (355) GNARL: subg_rfspy 2.2
D (365) GNARL: radio version 1.2
D (365) GNARL: frequency set to 868000000 Hz
E (365) gpio: gpio_isr_handler_add(476): GPIO isr service is not installed, call gpio_install_isr_service() first

Thank you

ecc1, 2022-11-27:
I've updated the main branch to call gpio_install_isr_service() which should fix this issue.

closed 2022-11-27 by ecc1