ecc1/gnarl #25: Unable to connect with pump

rasinhox1337, 2023-04-29:
Hi, 
i have pump medtronic 722WW. I tried to connect, but pump doesn't react. I build mmtune to check frequency.
I changed MMTUNE_START in pump_config.h to 868000000 and after scanning i have "Best frequency 0Hz". Every attempts end "wakeup failed". I set remote control in the pump settings. What am i doing wrong? Or what should i do to connect with pump.

ecc1, 2023-04-29:
What kind of ESP32/LoRa module are you using, and what antenna do you have connected to it?

rasinhox1337, 2023-04-29:
I  use [link](https://www.aliexpress.com/item/1005003090411881.html?pdp_npi=2%40dis%21PLN%21z%C5%82+97.37%21z%C5%82+97.37%21%21%21%21%21%40211b801716827738208017474e9006%2112000024021543125%21btf&_t=pvid%3A496444be-6837-4f4b-b17b-25fca6974944&afTraceInfo=1005003090411881__pc__pcBridgePPC__xxxxxx__1682773821&spm=a2g0o.ppclist.product.mainProduct&gatewayAdapt=glo2deu)

[image]

ecc1, 2023-04-29:
The README says in bold, italic letters that Version 2 modules are not supported.

closed 2023-04-29 by ecc1